woheller69/solxpect #3: Idea: Sum of all locations

marku5, 2023-04-18:
I've 6 different roofareas/location -> The sum of all genereated kWh have to be calculated separately/manually.

Idea: New "block": Sum of alle area-kWh-entries.

woheller69, 2023-04-18:
is already on the roadmap #1

closed 2023-04-18 by woheller69

marku5, 2023-04-18:
Ups...didn't see this line...

woheller69, 2023-04-18:
You can try this first implementation which sums up values of locations with same latitude/longitude
You need to switch on showSum in settings.

The city names should be like 
myPV | part1
myPV | part2
etc. In the sum it will take the name of one of the elements left of "|" -> myPV

Remove .zip and install
[app-release.apk.zip](https://github.com/woheller69/solxpect/files/11260773/app-release.apk.zip)

It always updates all locations with same lat/lon  and sums up the values so refreshing data may take some time as there are a lot of calculations to be performed...