woheller69/solxpect #4: No internet connection when on wifi

therealstretch, 2023-04-20:
When connected to wifi, pressing the refresh button
the toast "Das Gerät ist nicht mit dem Internet verbunden!"
appears and data is not refreshed.

When connected to mobile data (LTE) everything works as
expected.

All other applications on the phone do not have any issues
with the wifi-connection.

 - App version: 1.4
 - Android version: 12
 - Device: Redmi Note 9 Pro

woheller69, 2023-04-20:
probably an issue with a vpn or firewall.

I am using isReachable()
https://developer.android.com/reference/java/net/InetAddress#isReachable(int)

It seems Pings are blocked.

woheller69, 2023-04-20:
Can you please try this? I am trying a different way to find if a device is online.
This problem sometimes also happens in my weather apps for some users.
So maybe this is a solution:

Remove .zip from filename and install.
[app-release.apk.zip](https://github.com/woheller69/solxpect/files/11283461/app-release.apk.zip)

therealstretch, 2023-04-20:
Yes, this is working. Thank you!

woheller69, 2023-04-20:
another small change in the new online check. Hope it still works.
[app-release.apk.zip](https://github.com/woheller69/solxpect/files/11284122/app-release.apk.zip)

therealstretch, 2023-04-20:
Yes, still working.

DECALIOC, 2023-05-01:
I can confirm issues with the app "Firewall ohne Root". SolXpect is displayed as "uninstalled".

> probably an issue with a vpn or firewall.
> 
> I am using isReachable() https://developer.android.com/reference/java/net/InetAddress#isReachable(int)
> 
> It seems Pings are blocked.

woheller69, 2023-05-01:
Does it work with the version above?

DECALIOC, 2023-05-01:
Thank you! I just realized I still was on version 1.3 from F-Droid.

I updated to 1.4. The "uninstalled" flag is gone now.

However, I still see the message about solXpect being unable to connect to the Internet; both via LTE and WIFI, when the firewall is up.

woheller69, 2023-05-01:
Then you need to check why Firewall ohne Root blocks it and create a rule there

DECALIOC, 2023-05-01:
"Firewall ohne Root"'s log says the connection was allowed. Maybe the connection works, but solXpect just thinks it cannot connect.

I will observe, if the data are refreshed, for a few days.

Thank you!

woheller69, 2023-05-01:
If solXpect says there is no Internet connection it will not work. Then a check has been blocked by the firewall

closed 2023-05-02 by woheller69