woheller69/solxpect #10: Suggestions: Option to let stay last hour(s)

Mintulix, 2023-06-26:
Hi, great app, thank you!
I'm just trying out to tune the data that it fits to my PV and I have two little problems:

First, when I get my real data from the server for the last hour, the prediction of that time is gone, so I always have to make screenshots from the prediction to adjust the app data to the real world. That would be easier if there was an option to show the last few hours also.

Second, the way to change the shadowing is a bit inconvenient. There are 7 trees and a chimney I want to factor in.
Go to the prediction screen, click on the sun symbol, read the actual azimut and elevation, click ok, go to the location settings, find the entry and change it (and so on). It would be much easier, if the actual location of the sun would be available directly from the shadowing table. I could of course first notice all the data from one day at a seperate location and put them into the app later, but when I do it directly, I have a better control if the changed data fits.

woheller69, 2023-06-26:
you can switch on debug. Then you will also see past values for the current day.

Will add current sun position above the shading table.

Mintulix, 2023-06-26:
Great! Thank you very much!

woheller69, 2023-06-26:
You can try it here.

Backup your database
Uninstall
Remove .zip and install
Restore database

[app-release.apk.zip](https://github.com/woheller69/solxpect/files/11867388/app-release.apk.zip)

Mintulix, 2023-06-26:
Wow, that's fast! Looks good except that now "Speichern, Schliessen, Klonen" is vertical aligned instead of horizontal like before (and you have to scroll between them). Android 8.1 (no more updates available)

woheller69, 2023-06-26:
I only modified "Ändern" to "Speichern" 
Maybe your display does not have enough space for these 3 extra characters?

Mintulix, 2023-06-26:
That's how it looks like. I don't know if that affects many users...

[image]

You can scroll both sections seperately, the table and the menu.

woheller69, 2023-06-26:
I could change "Schliessen" to "Beenden" which is shorter...

closed 2023-07-15 by woheller69