woheller69/solxpect #13: Voluntary Turkish Translation

SirCrownguard, 2023-09-01:
Hello. I would like to make a voluntary Turkish translation for your application. Is there a language file that you can give me because I don't know about programming?

woheller69, 2023-09-01:
Sure. 

https://github.com/woheller69/solxpect/blob/main/app/src/main/res/values/strings.xml

Translate all strings, except those which are marked as translatable="false"

woheller69, 2023-09-01:
And there is a built-in help which should also be translated

https://github.com/woheller69/solxpect/blob/main/app/src/main/assets/help/help-en.html

SirCrownguard, 2023-09-01:
Since I couldn't see any line marked as translatable="false", I translated all lines. Both files are finished.

If there are new strings that need to be translated with a new update or if you want a Turkish translation for a project, you can tag me or contact me at the following e-mail address: departure_impromptu788@aleeas.com
[help-en.txt](https://github.com/woheller69/solxpect/files/12501988/help-en.txt)
[strings.txt](https://github.com/woheller69/solxpect/files/12501989/strings.txt)

woheller69, 2023-09-02:
Please try and check if help file works and if strings are correct.

Uninstall
Remove .zip and install

[app-release.apk.zip](https://github.com/woheller69/solxpect/files/12503275/app-release.apk.zip)

SirCrownguard, 2023-09-02:
Sorry for keeping you waiting.

Everything is working correctly. I saw that only 2 strings didn't work but they are strings not found in the strings.xml file you gave me. So I guess there is no problem.

One of the two strings I mentioned is the bottom option in the navigation bar. The second string is the Debug option in the Settings section.  If you want to translate, the first means "GitHub'da yıldız ver", the second means "Hata Ayıklama".

Thank you for taking your time. I was happy to assist you. I always like to help with translation.

woheller69, 2023-09-03:
Thanks! I have a few apps with Turkish translations but at least partly strings were translated by Google.
You could check if translations are good, e.g. omWeather, RadarWeather, or SmartEggTimer.
LavSeeker or GpsCockpit might also be interesting for new translations.

Here is a list of my apps:

https://search.f-droid.org/?q=woheller69

SirCrownguard, 2023-09-03:
I also saw that the descriptions of your applications are in English. Can I translate them too?
[image]

woheller69, 2023-09-03:
Yes, they are in fastlane directory

closed 2023-09-15 by woheller69