woheller69/solxpect #14: Add max/min temperatures

topcaser, 2023-11-06:
Hi, I am using your app regularly to double check how much solar energy is forecasted. This is very helpful to judge whether to switch on or not the radiators e. g during the intermediate summer/winter time (highly insulated house which can safely bridge the one or another day). Anyway it would be helpful to see the min and max temperatures per day to judge how long I can bridge it without switching on the radiators. Do you think, this is feasible.

Thanks for looking into it.

woheller69, 2023-11-06:
you can use my omWeather app to see that data

topcaser, 2023-12-01:
Yes, I use it indeed as well. But it would be so handy if everything is in one app,even I am pretty sure that you have the data already...

closed 2023-12-11 by woheller69