woheller69/solxpect #16: negative values today

Mintulix, 2024-02-15:
I never saw this before - the evening and night calculated values are negative (sorry for the huge picture):

[image]

I also use https://github.com/gvzdus/PVPrognose within iobroker, which uses more or less your code - also negative values today.
Is this a rare phenomenon or a real bug or...

woheller69, 2024-02-15:
[image]
looks like a bug in Open-Meteo API

Mintulix, 2024-02-15:
Ah, sorry, I didn't check this. I thought the first row with "0" values came from the API and the rest was calculated. It seems a fix is on the way, so this was quick! Thank you, I'll close this issue.

closed 2024-02-15 by Mintulix