woheller69/solxpect #17: Clarify the "Temperature coefficient" parameter

marcvangend, 2024-03-02:
**Describe the bug**
The [technical datasheet](https://www.solarwatt.de/canto/download/ehll9869mt54v7acc4ccodo254) of my panels specify three different temperature coefficients: Temperature coefficient Pmax (-0.31 %/K), Temperature coefficient Voc (-0.26  %/K) and Temperature coefficient Isc (0.04 %/K). The documentation about the "Temperature coefficient" parameter does not tell me which one I should use.

**Expected behavior**
I would appreciate if the docs would mention which temperature coefficient value to use when multiple values are given.

**Environment**
 - App version: 2.2
 - Android version: 14
 - Device: Google Pixel 7a

marku5, 2024-03-02:
It's Voc.

woheller69, 2024-03-02:
I would use the power coefficient.I don't think open circuit voltage or short circuit current are relevant.

marku5, 2024-03-02:
Ok, thanks for your correction.

closed 2024-03-28 by woheller69