woheller69/solxpect #19: Displaying current calculated PV output power

sschusser, 2024-03-09:
For those who are interested in using this app to forecast the output for an old fashioned PV system without a build in forecast:
In order to set and proof the parameters for calculation properly, it is of interest to compare the actual output power of the PV system with the calculated output power.
Hence I am asking whether it is possible to show the calculated expected output power value somewhere in the GUI. Not only the integrated value in terms of Wh for the upcoming hour.

woheller69, 2024-03-09:
if you have 200Wh the average power is 200W during that hour

sschusser, 2024-03-09:
If it is that easy, I can do this simple calculation. So, just one data
point per hour? I was expecting a better temporal resolution and, thus, the
total amount of energy displayed on the frontend to be an average value
calculated by some fancier routines: Integration over time axis.

Many thanks for the reply.

woheller69, 2024-03-09:
Open-Meteo already delivers cumulated values for each hour

sschusser, 2024-03-10:
I am getting confused about the actual situation.

if the displayed energy value for each hour is already summed up over
several values, I am not able to calculate the current expected power by
just dividing the energy value by the time delta of one hour. This just
gives a mean value for this time period.

That is why I was asking for displaying the current expected power.

Best regards

woheller69, 2024-03-10:
exactly, as I said you get the average power in that hour

closed 2024-03-28 by woheller69