woheller69/solxpect #20: Feature Request: Monthly and Yearly kWh

CalliOSM, 2024-04-26:
ProjectionsDescription: Hello SolXpect Team,
I've been using SolXpect and really appreciate how it helps users understand and predict daily solar irradiance.

I would like to propose an enhancement to the current functionality: the addition of monthly and yearly expected kWh projections.

Details: Adding monthly and yearly projections would provide users with a broader view of their potential energy generation, helping with long-term planning and optimization of solar panel usage. This feature could calculate the expected kWh by aggregating daily forecasts over each month and year, possibly adjusting for seasonal variations and historical weather patterns.

Benefit: This feature would significantly enhance the app's utility for homeowners and businesses alike by aiding in financial planning and energy management.

Thank you for considering this addition. I look forward to your thoughts on this suggestion.

woheller69, 2024-04-26:
there are websites for such calculations.
This is not the purpose of this app

woheller69, 2024-04-26:
you can e.g. use this calculator: https://pvgis.com/

closed 2024-04-26 by woheller69

CalliOSM, 2024-04-27:
Thank You for considering my request! Keep up the great work on this App!