woheller69/solxpect #21: Improvement suggestion: Store some old day-values

marku5, 2024-05-05:
I'm using solXpect for my PV. As I usually check the last days yield the next day (via homeassistant) it would be nice to see the last (3?) days in solXpect as well to compare the estimations with results.
This would ease finetuning e.g. shading settings. Currently I need to store the estimations separately.

Please consider keeping some old values instead of deleting them if date changes. Tnx.

woheller69, 2024-05-05:
you get the current day if you switch on debug.
so you can compare at the end of the day

marku5, 2024-05-05:
Yes, I've seen then debug-option.
I would like to still see yesterdays estimation (maybe even 2 more days in the past) to compare this to yesterdays yield on the next day. So I do not need to check data every evening for which I do not have time..but every few days....

closed 2024-05-27 by woheller69