woheller69/solxpect #23: Hourly vaues should be reset for ever day

marku5, 2024-06-25:
Bug or idea?

Main screen shows three sections:
1. Hourly values
2. Daily values as values
3. Daily values as barchart

Values in 2. and 3. are reset every day.
The sum of kWh in 1. is summarized for all shown days => These values shall as well be set to zero in midnight.

woheller69, 2024-06-30:
the hourly sum helps finding the right pramaters by comparing with measured production values

marku5, 2024-06-30:
To clarify: I like the hourly sum, just not the "no value reset at midnight".

woheller69, 2024-07-01:
Please try this apk

-Backup your settings
-Uninstall
-Remove .zip and install
-Restore settings

Values are reset at midnight if not in debug mode

[app-release.apk.zip](https://github.com/user-attachments/files/16049399/app-release.apk.zip)

marku5, 2024-07-01:
Works! Thanks for the fix/new feature!

commit 2974b4b1 (2024-07-01): -reset cumulated values at midnight, unless in debug mode
-bugfix for weekforcast because values are for preceding hour
fixes #23
closed 2024-07-02 by woheller69