woheller69/solxpect #24: Inverter shared (two MPTT strings) for two solxpect locations

floppyflop, 2024-07-12:
I have an east-west setup where two strings (with separate MPPT) share one inverter. I have setup two locations in the app (because of the different azimuth), but the maximum power of my inverter is for the sum of both locations. As long as the maximum power is never reached there is no issue, but at midday the sum of both locations could hit the maximum of the inverter. I could not find a setting to cover this. Did I miss something or could you find a way to implement this feature?

Environment:
 - App version: 2.4
 - Android version: 14
 - Device: Moto g84 5G

woheller69, 2024-07-12:
Unfortunately this is not possible at the moment and it would require a lot of changes in the code to add that feature...

woheller69, 2024-07-13:
Please try this apk:
-Backup database
-Uninstall
-Remove .zip and install
-Restore database

[app-release.apk.zip](https://github.com/user-attachments/files/16200930/app-release.apk.zip)

You can now mark the inverter as central inverter. Then the sum of these inverters is applied on system level.

floppyflop, 2024-07-13:
Thanks, you found a solution! It seems to work ok, it's easy to see how much more panels I could use on my inverter on expected solar data (by tweeking my locations). I can imagine other users with multiple locations and less  but more than one inverters would like to select a specific inverter to the location. Some remarks: It is now showing the central inverter power in screen 5 times during refresh.  Settings are not saved in the backup. Documentation still to be done.

woheller69, 2024-07-13:
The notification will be removed. It is just for testing.

What do you mean with: "users with multiple locations and less but more than one inverters would like to select a specific inverter to the location"

floppyflop, 2024-07-13:
It is a suggestions for users with more than one inverter each with multiple strings, to be able to select which string/location belongs to which inverter, therefore a possible extension to my situation with only one inverter.

woheller69, 2024-07-13:
This would be complex. But in this case they hopefully have the same orientation within each string and can use the old method.

woheller69, 2024-07-13:
In addition to the old method you also could add an additional location at the same position with 0 W power and 0 sqm area and just define the inverter limit.
For the inverter efficiency the value defined with each location is applied.

floppyflop, 2024-07-14:
If I don't show the sum energy I see that the inverter limit is per
location(name), i.e per inverter, that's good, but when I show sum energy I
see central inverter limit as the sum value of per location central
inverter limit values, which is indeed ok, but energy sum seems limited
only by this value, not the per location inverter limits. Total sum should
be the sum of inverters/locations.

Met vriendelijke groeten,

Folbert

woheller69, 2024-07-14:
If you build the sum it is limited by the sum of inverters marked as central inverters

floppyflop, 2024-07-14:
Maybe I'm on a different thoughtline. If I have two inverters myPVI and
myPVII with each two strings part1 and part2 I think I would create four
locations myPVI | part1, myPVI | part2, myPVII | part1 and myPVII | part2.
MyPVI inverter maximum is the sum of the inverter maxima of MyPV1 | part1
and MyPVI | part2, likewise for MyPVII.  I think then the total sum should
be MIN(myPVI | part1 + myPVI | part2 , myPVI inverter maximum) + MIN(myPVII
| part1 + myPVII | part2 , myPVII inverter maximum), and not
MIN(myPVI | part1 + myPVI | part2 + myPVII | part1 + myPVII | part2 , myPVI
inverter maximum + myPVII inverter maximum).

Met vriendelijke groeten,

Folbert Pijper
tel.:     06-52539216
e-mail: ***@***.***

woheller69, 2024-07-14:
How many orientations do you have? I would do one location per orientation

floppyflop, 2024-07-14:
At the moment I have two orientations, one inverter with two MPPT strings.
The provided solution works fine for me in this case! The rest is just for
possible future expansions or thinking ahead of other peoples situation
with more than one inverter and multiple strings, or to see how things
would work out in different configurations, mainly if the inverter is
limiting during peak hours.

Met vriendelijke groeten,

Folbert Pijper
tel.:     06-52539216
e-mail: ***@***.***

commit 9ab548b9 (2024-07-17): V2.5
fixes #24
closed 2024-07-17 by woheller69