woheller69/solxpect #26: Suggestion: Tab for sums

jochen-01, 2024-08-05:
As with version 2.5 we can see sites (e.g. different parts of a roof) separate or as a sum. Please add a separate tab for sums to see each part AND the sum simultaneously. 

As an additional option let select different sites for different sums - for people with more than one roof. ;-)

jochen-01, 2024-10-30:
Hi,
re. my 2nd point: Assume a farm. You have 2 roof sides of the House and 2 roof sides on the barn. I meant sub sums of the house panels and another sub sum of the barn panels. But that's not a priority.
What I want to see are 3 tabs: 
<location> sum 
<location> south
<location> north
What I have now is one sum screen if I have enabled "show sums" and 2 tabs with the "south" and "north" with disabled "show sums".

jochen-01, 2024-10-30:
That does not help. If I modify one set of coordinates I won't have the chance to see the sum. 

To explain it: One roof, two Cardinal directions (south and north), both equipped with solar panels.
I want to see each set of solar panels (north / south) and a sum of them. All in separate tabs: (xxx north, xxx south, xxx sum).

jochen-01, 2024-10-30:
You're right. It is confusing. ;-)
Another setting like "show sum(s) for every location" should bring up an additional tab with the sum:
I.e.: 
xxx north
xxx west
xxx south
xxx sum
Something like that...

woheller69, 2024-10-30:
just keep the set as you have it right now. This will give the total of all.
Then ADD (not modify) additional locations with slightly different lat/lon and you will see these separately or if some of them have same lat/lon they will be grouped.

jochen-01, 2024-10-30:
Currently I have just one installation. One roof, with one panel string directed south and one panel string directed north.
For me it would be absolutely o.k. to have 3 tabs (south, north, sum of both). That is probably the setup most people are using.

closed 2024-11-01 by woheller69