woheller69/solxpect #27: Add forecast as widget on main screen

averter, 2024-09-30:
**Describe the bug**
Not a bug, more of a request so that the forecast information could be added as a widget on my smartphone main screen. 

**Expected behavior**
A customizable icon with information about the power/output in the following hours/days, similar with [other weather forecast apps](https://duckduckgo.com/?q=weather+forecast+app&t=lm&iar=images&iax=images&ia=images), such as omni jaws.

**Screenshots**
See above link.

**Environment (please complete the following information):**
 - App version: v2.5
 - Android version: LineageOS 21 (Android 14)
 - Device: Redmi Note 4X

achimghita, 2024-11-20:
How precise is the estimation (for today)? Did anyone compare an actual system production to the simulated one? Does it auto correct as the weather forecast becomes reality (for ex: the forecasted rain in the morning did not happen, and it's sunny now)

woheller69, 2024-11-20:
For me usually within 10%.
It gets updated when the weather forecast is updated

averter, 2024-11-23:
Sorry but why are you writing comments unrelated to what I wrote above?