woheller69/solxpect #28: Feature request

fschli, 2024-12-04:
I would love to be able to see the summed up energy, that will be produced from now until sunset. This could be implemented using a number or a two-colored column in the diagram.

Thank you for developing this useful app.

Alex-Esko, 2025-04-03:
You can easily calculate this value:

1. Enable debug, to get ∑ values.
2. The ∑ value is the summed up kWh from midnight to current hour.
3. The total production on this day is in the weekly tab.
4. Substract ∑ from total day-production.

[image]

woheller69, 2025-04-28:
Try this.

[app-release.apk.zip](https://github.com/user-attachments/files/19936984/app-release.apk.zip)

Backup
Uninstall
Remove .zip and install
Restore

schlifex, 2025-04-28:
Looks good. Thank you for the update.

woheller69, 2025-04-29:
->V2.6

closed 2025-04-29 by woheller69