woheller69/solxpect #29: Forecast history

achimghita, 2025-01-03:
Feature request: Allow scrolling through the chart both ways - in the past (cached data?) and in the future.

woheller69, 2025-01-04:
you can activate "Debug". Then you can see values starting 00:00 of the current day

Alex-Esko, 2025-04-03:
It would be great to have historical data of the last days in solXpect. So one can compare actual solar production with the estimate of solXpect and fine tune parameters like shading and "Diffuse radiation efficiency".

In my case, after tuning the parameters a few days, the estimate is still 25% off.

hanskaas, 2025-04-14:
Könnte man die Voraussage nicht in einem Graph zusätzlich oder alternativ zu den Stundendaten darstellen? Wäre zur schnelleren Information dienlich. Hintergrund  ist die KI, die immer mehr In die Steuerung der Einspeisung Einzug hält. Die Prognosen über die zu erwartenden Erträge werden für den betreffenden Tag in einem Graph dargestellt. So könnte man schneller den Vergleich zu solexecpt herstellen.