woheller69/solxpect #30: Error while backing up without an existing Documents directory

schlifex, 2025-04-28:
**Describe the bug**
If you try to store a backup, but the directory /storage/emulated/0/Documents doesn't exist, you get an error.

**How to Reproduce**
Steps to reproduce the behavior:
1. Sichern/Wiederherstellen
2. Click on 'Datenbank sichern'
3. Click on 'OK'
4. See error: net.lingala.zip4j.exception.ZipException: java.io.FIleNotFoundException: /storage/emulated/0/Documents/solXpect.zip

**Expected behavior**
Error message, that's understandable for everyone or ability to choose the backup directory.

**Environment (please complete the following information):**
 - App version: 2.5
 - Android version: 15
 - Device: Google Pixel 6

woheller69, 2025-04-28:
what is your OS?
I never had a device without this directory....

Will check and create if necessary in next version.

schlifex, 2025-04-28:
To be honest, I wouldn't be surprised if I deleted it at some point. ;)

Do not hesitate to close the issue if you do not consider it relevant.

woheller69, 2025-04-29:
fixed

closed 2025-04-29 by woheller69